reilleya/openMotor #225: replace appdirs with platformdirs

bdale, 2024-04-03:
In Debian bug #1067977, Simon McVittie points out that python3-appdirs is dead upstream and will not be included in the next Debian stable release.  He suggests migrating to python3-platformdirs, which is reportedly a fork of appdirs with a very similar API.

See the full but report at https://bugs.debian.org/1067977

benrussell11, 2024-04-03:
Hi Andrew,

 

Do you have date/target for the next release?

 

Ben

 

👍 Will do in the next release!

reilleya, 2024-04-05:
Hey Ben, no specific plans as of now. I don't get as much time to work on this as I used to, which is slowing things down. I'd need to decide exactly what the scope of 0.6.0 will be to have a better idea of when I could complete it. I don't think there's much room for additional simulation features without a rewrite of the application, so it'll mostly be focused on fixing issues like the interface not scaling well on windows machines with high DPI displays. Would be great to get it out this year as those machines are just becoming more and more common.

bdale, 2024-08-24:
Just a gentle ping.  Any update?

reilleya, 2024-08-25:
Thanks for the reminder! Done.

closed 2024-08-25 by reilleya

bdale, 2024-08-25:
Thank you!  FYI, I've applied this commit as a patch for the 0.5.0-3 version of the Debian package, and all seems fine.  Looking forward to 0.6 whenever that happens.