reilleya/openMotor #228: AttributeError: 'ArtistList' object has no attribute 'pop'

AlexKari, 2024-05-31:
Hi,

When running Arch Linux, I have the following error every time I try to edit a grain

`(.venv) [user@Kreation openMotor-0.5.0]$ micro uilib/widgets/grainPreviewWidget.py 
(.venv) [user@Kreation openMotor-0.5.0]$ python3 main.py 
0.0000 [LOG] ################################################################################
0.0000 [LOG] Application started at 2024-05-31T09:34:58.958954
0.5865 [LOG] Starting new motor file
0.5889 [LOG] Opening window
QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout
0.9226 [LOG] Window opened
9.1469 [ERR] Traceback (most recent call last):
9.1469 [ERR]   File "/home/user/Games/openMotor-0.5.0/uilib/widgets/grainPreviewWidget.py", line 50, in updateView
9.1469 [ERR]     self.ui.tabRegression.cleanup()
9.1469 [ERR]   File "/home/user/Games/openMotor-0.5.0/uilib/widgets/grainPreviewGraph.py", line 41, in cleanup
9.1469 [ERR]     self.plot.lines.pop(0)
9.1470 [ERR]     ^^^^^^^^^^^^^^^^^^^
9.1470 [ERR] AttributeError: 'ArtistList' object has no attribute 'pop'
9.1470 [ERR] 
`

reilleya, 2024-06-02:
This one is already fixed on `staging`, and will be in the eventual v0.6.0 release! See [this](https://github.com/reilleya/openMotor/issues/210) issue.

closed 2024-06-02 by reilleya

benrussell11, 2024-06-02:
Hi Andrew,

 

Maybe its time to do a production bug fix release.

 

Ben