reilleya/openMotor #232: Update pyyaml > 5.1.2

dhbarr, 2024-09-19:
https://nvd.nist.gov/vuln/detail/cve-2019-20477

> PyYAML 5.1 through 5.1.2 has insufficient restrictions on the load and load_all functions because of a class deserialization issue, e.g., Popen is a class in the subprocess module. NOTE: this issue exists because of an incomplete fix for CVE-2017-18342.

dhbarr, 2024-09-19:
Disregard, I see from `blob/staging/requirements.txt` that `PyYAML==6.0` -- do we need to do anything about old / existing install base?

reilleya, 2024-09-22:
Hmm, certainly not great to have a vulnerability like this in old versions of the software, but I'm not sure what I can do other than suggest that people upgrade, and there isn't a notification system in the application or anything.

closed 2024-09-22 by reilleya