reilleya/openMotor #238: [Discussion] Pre-built binaries for Linux

PixelShadow, 2025-04-29:
I run Linux an I tried to build OpenMotor from source. I ran into several issues regarding dependencies. I'll probably solve them but it still takes a lot of time. Is there any way we can get pre-built binaries of this software for Linux ?

radio-satellites, 2025-06-30:
Same for me. Am on Fedora and building is not fun.

radio-satellites, 2025-06-30:
Turns out I needed to update my requirements.txt to build, that helped a lot. To anyone looking:
[requirements.txt](https://github.com/user-attachments/files/20978289/requirements.txt)

reilleya, 2025-07-01:
Hi! Totally willing to provide packaged versions of releases for Linux, but I'm not sure where to start. What distros/formats should I target? Alternatively, willing to accept PRs that make it easier to run on Linux. Not sure if that requirements.txt file is an improvement, though, looks like the change was just removing the versions for a couple of packages. Is the issue that the versions specified don't have binaries available for Linux? Would prefer specifying versions that do (latest?) rather than dropping the version.

radio-satellites, 2025-07-01:
@PixelShadow Good overview. I don't think the process for FlatPak is that difficult - it's mostly creating a .yml manifest, then running flatpack-builder. To get it on FlatHub, I think you need to submit a pull request [here](https://github.com/flathub/flathub/pulls). I haven't done the process ever myself but do imagine it wouldn't be that difficult.

Digitalpeer1, 2025-07-11:
But flatpak makes it usable on more systems than debian and arch.

vahpr, 2025-08-26:
Forgive the newbie Linux user question - can openmotor be installed and run via Wine? 

Regardless, it would be great to have a package available.